phoddie/node-red-mcu #1: mqttclient manifest

sammachin, 2022-06-23:
Been trying to get a simple Node-RED flow running on a moddable two, 

I noticed an error trying to build, in the manifest.json for this project, https://github.com/phoddie/node-red-mcu/blob/7291f8b13b47c3faa9c5482e2c3b7bfc3bc8f141/manifest.json#L5

But in my examples the io/tcp/mqttclient project has its manifest file as manifest.json.

Is there some custom config you added to the mqttclient for node-red-mcu?

My flow builds but it doesn't seem to connect, I have an mqtt in node to subscribe to a top, a function node that does msg.payload*2 and then an mqtt out node to publish to a different topic.

phoddie, 2022-06-23:
Sorry for the trouble, @sammachin. The `manifest_mqttclient.json` is missing. We'll get the public Moddable SDK repository updated with that. In the meantime, here's the content of the manifest.

<details>
<summary>manifest_mqttclient.json</summary>

```json
{
	"modules": {
		"*": [ 
			"./config",
			"$(MODULES)/network/dns/moddnsparser"
		],
		"embedded:network/dns/resolver/udp": "../../udp/dns/dns",
		"dns": "$(MODULES)/network/dns/dns",
		"dns/parser": "$(MODULES)/network/dns/dnsparser",
		"dns/serializer": "$(MODULES)/network/dns/dnsserializer",
		"embedded:network/mqtt/client": "./mqttclient"
	},
	"preload": [
		"config",
		"embedded:network/dns/resolver",
		"dns",
		"dns/parser",
		"dns/serializer",
		"embedded:network/mqtt/client"
	]
}
```

</details>

>  it doesn't seem to connect

If you replaced manifest_mqttclient.json with manifest.json it may have overridden the project main.js which would explain why nothing happened. But, the network connection also needs to be configured.  If you are running on the simulator on macOS, it will automatically use your computer's network connection:

```
mcconfig -d -m
```

If you are running on an ESP32, you need to provide the Wi-Fi credentials so that the ESP32 can connect:

```
mcconfig -d -m ssid="My SSID" password="My Wi-Fi password"
```
I just tested a simple flow with MQTT in and out successfully (with the manifest added above).  FWIW, here's the test flow:

<details>
<summary>mqtt test</summary>

```json
[
    {
        "id": "6d1fc1f5b13fb5b7",
        "type": "tab",
        "label": "mqtt test",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "00470d78886101aa",
        "type": "inject",
        "z": "6d1fc1f5b13fb5b7",
        "name": "inject start",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "5",
        "crontab": "",
        "once": true,
        "onceDelay": "0",
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 154,
        "y": 120,
        "wires": [
            [
                "962842be115088ab"
            ]
        ]
    },
    {
        "id": "f1d0832b756d8526",
        "type": "debug",
        "z": "6d1fc1f5b13fb5b7",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": true,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 410,
        "y": 200,
        "wires": []
    },
    {
        "id": "f832e9b85c4d9be4",
        "type": "mqtt in",
        "z": "6d1fc1f5b13fb5b7",
        "name": "",
        "topic": "/moddable/test/in",
        "qos": "0",
        "datatype": "utf8",
        "broker": "40068d0db5e8893c",
        "nl": false,
        "rap": true,
        "rh": 0,
        "inputs": 0,
        "x": 170,
        "y": 200,
        "wires": [
            [
                "f1d0832b756d8526"
            ]
        ]
    },
    {
        "id": "962842be115088ab",
        "type": "mqtt out",
        "z": "6d1fc1f5b13fb5b7",
        "name": "",
        "topic": "/moddable/test/out",
        "qos": "0",
        "retain": "",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "40068d0db5e8893c",
        "x": 370,
        "y": 120,
        "wires": []
    },
    {
        "id": "40068d0db5e8893c",
        "type": "mqtt-broker",
        "name": "mosquitto.org",
        "broker": "test.mosquitto.org",
        "port": "1883",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "sessionExpiry": ""
    }
]
```

</details>

phoddie, 2022-06-24:
Just FYI – [`manifest_mqttclient.json`](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/io/tcp/mqttclient/manifest_mqttclient.json) is in the Moddable SDK repository now.

mtoko, 2022-06-25:
Thanks for this, I have got the MQTT working using your sample flow and also with some other flows which include GPIO out pins. The only issue is that it will only connect to the broker in your example. I have tried a few other brokers, both cloud and local but none of these work even without authentication. When you copy a flow from one instance of Node Red to another it does not copy the credentials, could you add these into the flow after you have copied this into the node.js file?

phoddie, 2022-06-25:
>  I have got the MQTT working using your sample flow and also with some other flows which include GPIO out pin

Very nice!

> The only issue is that it will only connect to the broker in your example. I have tried a few other brokers, both cloud and local but none of these work even without authentication. When you copy a [...]

I would be happy to look into this, but I'm not sure I completely understand. Would you open a new issue and provide a sample flow to reproduce the problem?

closed 2022-06-28 by phoddie