phoddie/node-red-mcu #2: MQTT connection and Authorisation

mtoko, 2022-06-25:
I used the example MQTT flow from the earlier issue as a starting point and after rebuilding the SDK after manifest_mqttclient.json was added this worked.
I made a simple flow that was connected to an MQTT in and operated the built in LED on an ESP32 and sent out an MQTT out when this received a message. This flow worked, it was possible to publish a message to the broker and operate the LED 
Flow dowloaded to ESP32

[Flow downloaded into ESP32 connected to test Mosquitto Broker.txt](https://github.com/phoddie/node-red-mcu/files/8985150/Flow.downloaded.into.ESP32.connected.to.test.Mosquitto.Broker.txt)

Test flow on Node Red to operate above

[Flow to send MQTT to test Mosquitto fom Node Red.txt](https://github.com/phoddie/node-red-mcu/files/8985151/Flow.to.send.MQTT.to.test.Mosquitto.fom.Node.Red.txt)

I created a Node Red flow on an instance of Node Red on the same network using an MQTT Broker node without authorisation and then used the same flow above with the broker credentials changed, also no authorisation. The ESP did not connect to the broker and the broker was tested with other publish/subscribe messages which worked. This is the flow with the broker
[Node Red Flow with MQTT broker node to create local MQTT without Auth.txt](https://github.com/phoddie/node-red-mcu/files/8985132/Node.Red.Flow.with.MQTT.broker.node.to.create.local.MQTT.without.Auth.txt)

The first issue is that the ESP will not connect to another broker on the same network even without authorisation.
In normal circumstances the MQTT would use user and password for auth, but if you copy flows from one NR instance to another these credentials are not copied and need to be re-entered in the copied flow config. If a flow is copied to the Node.js folder, how can the user and password be included?

phoddie, 2022-06-25:
Thank you for writing up this detailed report. I only have time for a quick look now. The "..without auth" flow uses the "aedes broker" instead of the built-in "mqtt-broker" and "mqtt-in" (which is used by the first example). The MCU version doesn't implement "aedes broker", only "mutt-broker". Is it possible to use the built-in "mqtt-broker" and "mqtt-in" here instead and, if so, does it make a difference?

phoddie, 2022-06-25:
There are many kinds of nodes in Node-RED and not all can be supported on an MCU, it now traces an error message on unrecognized node types. With the "...without Auth" flow it now traces `Unsupported Node "aedes broker"`.

mtoko, 2022-06-25:
Apologies if I have not been clear, I was attempting to describe the tests I have done.
Summary
The ESP can communicate with the test.mosquitto broker that you have used in your example and also in my flow.
I have created a broker in a full instance of Node Red (not on the MCU) by using the Aedes node. The purpose was to test the communication with a local broker on the same network and not in the cloud. Both of these tests would not require auth with the broker. So, the first issue is why would the MCU not connect to a local broker without auth but will connect to a cloud broker without auth.
The second issue is that it is expected to use auth to connect to a broker usually using a user and password. When copying a flow from one instance to another these auth credentials are not copied and have to be re-entered. I think these credentials are stored in an encrypted file in the Node Red file structure. If the intention is to use MQTT on the MCU with auth then there would need to be a method to add the credentials to the flow after it is copied into the Node.js file and then download into the MCU. 
Hope this is clear, please ask reply with any questions or test that I can try

Steve-Mcl, 2022-06-25:
@mtoko I believe this is a "feature" of node red in that credentials are not exported.  However, if you build your flow and enter credentials in a normal node red editor and immediately export before deploying. Then it should contain the credentials you entered.  Give it a call and inspect the JSON. You should see username and password. 
If that doesn't work, you could try posting the flow via the admin endpoints of node red (see node-red API documentation) ( this of course assumes the admin endpoints are still available in this iteration)

mtoko, 2022-06-25:
@Steve-Mcl Great suggestion! I have configured the MQTT nodes with a cloud broker and used a user/password auth and it works!! When you inspecxt the exported flow you can see an object at the bottom of the MQTT node with the credentials.This is a trick to remember for the future. Thanks
@phoddie If you are happy with this solution you could close this issue?

phoddie, 2022-06-25:
@mtoko & @Steve-Mcl, thank you both for sorting this out (and @mtoko for trying out this work-in-progress!).

closed 2022-06-25 by phoddie