phoddie/node-red-mcu #3: Testing Nodes

mtoko, 2022-06-26:
I have been using a simple method to test Node Red nodes to begin to identify those that are working. The method is to create a simple flow with limited nodes focused on the node on test. The first results are that when a node does not work the error has been the same. I have concentrated on some of the core nodes but when testing a DS1820 node from the palette, this produced the same error. If this can be resolved it could fix the issues with a large number of nodes and get them working.
The screenshot of the error is
[image]
The whole debug output text is
[TestedNodesDebug.txt](https://github.com/phoddie/node-red-mcu/files/8986138/TestedNodesDebug.txt)

phoddie, 2022-06-26:
That exception means that you are using a node that is not implemented. The message is not obvious. ;) Yesterday I committed a change to display the name of the missing node instead. If you update your copy of this repository, you should at least get a better error message.

I tried to install a DS1820 Node from Node-RED > Settings > Palette > Install. Unfortunately, the search didn't show DS1820 available. The Moddable SDK does support the DS18X20 OneWire temperature sensors thanks to the work of @wilberforce. If you can explain how to install the DS1820 Node to the Palette, perhaps I can find a way to get that working. Thank you.

mtoko, 2022-06-26:
Thank you, I understand that each node needs to be specifically implemented. Is this a difficult process or with enough training could a trained monkey do it (no, disrepect to monkeys) ;-)
I have done a video  to show how to install the DS18B20 nodes. There are a few of them so I have chosen the one that fits with the working Pi nodes.

https://user-images.githubusercontent.com/4711179/175819175-8c10f7ee-8e5f-40f9-8f73-44c430ef7946.mp4

mtoko, 2022-06-26:
With two sensors connected

https://user-images.githubusercontent.com/4711179/175820624-742345f9-47a5-4fea-9e65-6d01714137ea.mp4

phoddie, 2022-06-26:
Very helpful, thank you!

phoddie, 2022-06-27:
Quick update: this looks very possible to implement. Unfortunately, I can't seem to find my DS1820 sensors. I ordered some more. They should arrive in the next few days.

mtoko, 2022-06-27:
Thanks for the update, at last count there were over 4000 user contributed nodes for Node Red. Is there any way that I could help with implementation?

phoddie, 2022-06-28:
> at last count there were over 4000 user contributed nodes

Yes. ;) I'm pretty sure not all of them will work on an MCU. I think many could. The biggest issues is that they are written for Node.js and we're not emulating that. The Nodes will have to be adapted or re-written.

> Is there any way that I could help with implementation?

Please! From my experience thus far, implementing a Node requires three skills: knowledge of (1) how Node-RED works, (2) JavaScript, and (3) Ecma-419 and/or the Moddable SDK. I definitely do not have (1), but I'm trying to learn. Maybe a good place to start is helping to fill in some of the core Nodes? That could be a good way to learn how the pieces fit together. The [Range node](https://github.com/phoddie/node-red-mcu#range) has a couple of unimplemented features, for example, that might be straightforward to complete. The readme on this repository lists other unimplemented bits.

FWIW – I've made some good progress on a subset the HTTP Request. I hope to have that committed tomorrow.

mtoko, 2022-06-28:
Thanks for this info, I think that my skills are lacking in all three, but I am working to learn. I have looked at the code in nodered.js and am working on a websocket in node which is similar to the MQTT in node to try and understand how this goes together.
From my limited understanding, flows can only contain nodes that are already in the nodered.js? If this is the case, then would it not be necessary eventually to have some kind of "palette" (like Node Red) so that only nodes that are in your flow are included 
I am using the Javascript section of your book as a reference ;-)
I will also check the features in the Range node.
I did mention this project on the Node Red Forum and there was interest, excitement, and some disbelief

phoddie, 2022-07-01:
The DS18B20 node is now implemented. I tested it with two temperature sensors on ESP32 (connects to GPIO 13) and macOS (which uses simulated sensors). It should also work on ESP8266 (pin 13) but I didn't try that yet.

Thank you again for the videos. That really helped. 

Here's a simple flow I used for testing.

<details>
<summary>DS18B20 flow</summary>

```json
[
    {
        "id": "91718ed2c8bd1088",
        "type": "tab",
        "label": "ds18b20",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "905abf6548d71dc1",
        "type": "inject",
        "z": "91718ed2c8bd1088",
        "name": "",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "2",
        "crontab": "",
        "once": true,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 190,
        "y": 160,
        "wires": [
            [
                "da3e45b40fc31666"
            ]
        ]
    },
    {
        "id": "da3e45b40fc31666",
        "type": "rpi-ds18b20",
        "z": "91718ed2c8bd1088",
        "topic": "",
        "array": true,
        "name": "",
        "x": 390,
        "y": 160,
        "wires": [
            [
                "acf7c93d5aabe5e2"
            ]
        ]
    },
    {
        "id": "acf7c93d5aabe5e2",
        "type": "debug",
        "z": "91718ed2c8bd1088",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": true,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 620,
        "y": 160,
        "wires": []
    }
]
```

</details>

mtoko, 2022-07-01:
Great! Thanks for this, it is not only useful for practical implementation of these common sensors, but the code provides an example of how other sensors could be implemented.

Just a side note
Raspberry Pi pico W (wireless) was available for purchase yesterday

phoddie, 2022-07-01:
>  the code provides an example of how other sensors could be implemented.

Yes, I hope so. As part of supporting that, the DS18B20  is implemented as a separate plug-in module rather than as part of the main Node-RED MCU code. 

I'd like to implement a sensor using the [Ecma-419 Sensor Class Pattern](https://419.ecma-international.org/#-13-sensor-class-pattern). That is a general purpose API for integrating sensors. That would make it straightforward to integrate future sensors. We have a growing suite of [Ecma-419 compatible sensor classes](https://github.com/Moddable-OpenSource/moddable/tree/public/modules/drivers/sensors) in the Moddable SDK for that. Do you know if there are corresponding Node-RED nodes for any of those already?

mtoko, 2022-07-01:
These are the sensors that I have been able to find after a quick search. It is important to remember that many of these are contributed by members of the community and having different levels of support. At the least they may have worked in the past and can be a starting point. I use plenty of BME280 which is similar to BMP280

In addition I have seen some flows where the i2c nodes are used and the data is read and then configured in the flow so this could be a "universal" solution for some i2c devices but would depend on i2c experience and device data 
[image]

mtoko, 2022-07-01:
I have looked at the plugin for the Ds18b20 and this is a sensible approach for the future. It would not be feasible to put all the nodes in main Node-RED MCU code. This should only contain the core nodes similar to the standard Node Red install.
In order to optimise in the future would it be possible to scan the flows.js file and then build a custom nodered.js file in some type of prebuild? Thanks for updating all the instructions for the new plugin approach, I will need to pull all these changes to test

phoddie, 2022-07-02:
Thanks for the list. The BME280/BMP280 seems like a good starting point. It is common, not too complicated, and we have an Ecma-419 Sensor Class for it.

> In addition I have seen some flows where the i2c nodes are used and the data is read and then configured in the flow so this could be a "universal" solution for some i2c devices but would depend on i2c experience and device data

We do that kind of thing often (it is how the BMP280 driver is implemented). So, that's an option. I think there are several I2C APIs for Raspberry Pi which complicates things, or is one commonly used in Node-RED?

> In order to optimise in the future would it be possible to scan the flows.js file and then build a custom nodered.js file in some type of prebuild?

Yes, something like that. I've mentioned a couple times that I want to change how the JSON is converted. That can be an off-device step and it can optimize several things, including which modules are included. That's important because some devices (ESP8266, Raspberry Pi Pico) have pretty limited flash storage space.

phoddie, 2022-07-26:
Using the sensor node, it is now possible to work with many different sensors. The [readme](https://github.com/phoddie/node-red-mcu/blob/main/nodes/sensor/readme.md) explains some details.

closed 2022-07-26 by phoddie