phoddie/node-red-mcu #4: module "system" not found

Steve-Mcl, 2022-06-26:
Hi 

I was playing with this to get a feel for it but cannot get any further than `module "system" not found` when attempting to run this in the simulator on node-red (on windows box)

executing `mcconfig -d -m` results in...
```
win_xs.c
xsAll.c
xsAPI.c
xsArguments.c
xsArray.c
xsAtomics.c
xsBigInt.c
xsBoolean.c
xsCode.c
xsCommon.c
xsDataView.c
xsDate.c
xsDebug.c
xsError.c
xsFunction.c
xsGenerator.c
xsGlobal.c
xsJSON.c
xsLexical.c
xsMapSet.c
xsMarshall.c
xsMath.c
xsMemory.c
xsModule.c
xsNumber.c
xsObject.c
xsPlatforms.c
xsProfile.c
xsPromise.c
xsProperty.c
xsProxy.c
xsRegExp.c
xsRun.c
xsScope.c
xsScript.c
xsSourceMap.c
xsString.c
xsSymbol.c
xsSyntaxical.c
xsTree.c
xsType.c
xsdtoa.c
xsmc.c
xsre.c
# xsc screen.xsb
# xsc time.xsb
# xsc timer.xsb
# xsc Resource.xsb
# xsc wifi.xsb
# xsc socket.xsb
# xsc net.xsb
# xsc sntp.xsb
# xsc modules.xsb
# xsc crypt.xsb
# xsc base64.xsb
# xsc logical.xsb
# xsc mqtt.xsb
# xsc mqttserver.xsb
# xsc websocket.xsb
# xsc config.xsb
# xsc main.xsb
# xsc nodered.xsb
# xsc nodes.xsb
# xsc ~.embedded\network\dns\resolver\udp.xsb
# xsc dns.xsb
# xsc dns\parser.xsb
# xsc dns\serializer.xsb
# xsc ~.embedded\network\mqtt\client.xsb
# xsc mc\config.xsb
# xsid screen.c.xsi
# xsid modTime.c.xsi
# xsid timer.c.xsi
# xsid modTimer.c.xsi
# xsid Resource.c.xsi
# xsid modInstrumentation.c.xsi
# xsid modSocket.c.xsi
# xsid modNet.c.xsi
# xsid modResolve.c.xsi
# xsid modsntp.c.xsi
# xsid modules.c.xsi
# xsid modCrypt.c.xsi
# xsid chacha.c.xsi
# xsid fips180.c.xsi
# xsid fips197.c.xsi
# xsid fips46.c.xsi
# xsid ghash.c.xsi
# xsid rc.c.xsi
# xsid rfc1321.c.xsi
# xsid modBase64.c.xsi
# xsid modLogical.c.xsi
# xsid moddnsparser.c.xsi
# xsid screen.h.xsi
# xsid modTimer.h.xsi
# xsid modInstrumentation.h.xsi
# xsid modSocket.h.xsi
# xsid chacha.h.xsi
# xsid fips180.h.xsi
# xsid fips197.h.xsi
# xsid fips46.h.xsi
# xsid ghash.h.xsi
# xsid kcl.h.xsi
# xsid kcl_symmetric.h.xsi
# xsid rc.h.xsi
# xsid rfc1321.h.xsi
# xsl modules
# exception: (host): module "system" not found!
### ReferenceError: (host): module "system" not found!
NMAKE : fatal error U1077: 'C:\Users\Stephen\repos\github\moddable\build\bin\win\debug\xsl.EXE' : return code '0xd'
Stop.
```

I can however run other examples (e.g. `examples\helloworld` , `network\mqtt\mqttbasic` and `examples\piu\balls` all work on simulator on windows)

Any ideas?

phoddie, 2022-06-26:
@Steve-Mcl, thanks for giving it a try. Unfortunately, the Windows simulator is not currently supported. As the [readme](https://github.com/phoddie/node-red-mcu#running-flows-on-an-mcu) notes:

> Flows run on ESP8266 and ESP32 MCUs, and in the Moddable SDK simulator on macOS computers

The MCUs can be quite inexpensive: basic modules are usually well under $10. Here's an [ESP32](https://www.amazon.com/Development-Microcontroller-Integrated-Antenna-Amplifiers/dp/B09GK74F7N/) bundle and a single [ESP8266](https://www.amazon.com/HiLetgo-Internet-Development-Wireless-Micropython/dp/B010O1G1ES).

FWIW – Windows simulator support is possible and we will get there. What's needed in this case is an ECMA-419 TCP socket implementation for Windows (which is what the MQTT support is built on).

I will make a note to add a better error message for unsupported simulator targets, so that others can more quickly understand what is going on here. Apologies for the trouble.

Steve-Mcl, 2022-06-26:
@phoddie thanks for the quick feedback.  I scanned the readme several times but missed that particular line

No worries, I'm sure you'll get there with the windows support (I'd be happy with Windows or Linux simulator support TBH - I just dont/wont do "apple")

Also, I do have a spare ESP8266 knocking around - now to hunt down the relevant moddable docs on how to do it with one of those.

phoddie, 2022-06-26:
> Also, I do have a spare ESP8266 knocking around - now to hunt down the relevant moddable docs on how to do it with one of those

Perfect. Since you already have the Moddable SDK installed on your Windows machine, you just need to add the ESP8266 tools. The details for that are [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp8266.md#windows).

closed 2022-06-28 by phoddie