phoddie/node-red-mcu #14: Function node external modules

spillz-dxb, 2022-07-15:
Hello!
Been trying out node-red-mcu , smaller flows doesn't  take up resources.
One thing before i try , does node-red-mcu support using function nodes with external modules .
If possible one could just use scripts for rest of the npm modules ,without the need to add each node manually.
Next is trying out servers and listeners !

Good job peter and the team at node-red

spillz-dxb, 2022-07-15:
Thanks peter 
> Separately, the [Compatibility Node](https://github.com/phoddie/node-red-mcu#compatibility-node) is a first attempt to implement support for running existing Node-RED modules. It works for the "lowercase" example and should work for many modules implemented without external dependencies.

Have tried the lower-case example, it works however the function node with external npm modules failed .
I am trying to get [https://www.npmjs.com/package/node-red-contrib-bacnet](url) and [https://flows.nodered.org/node/node-red-contrib-modbus](url) to get data in .

Would you be able to point me in the right direction please?

> > Next is trying out servers and listeners !
> 
> Those aren't implemented yet. They are on the roadmap.

Awaiting some inputs on the same

Glad to contribute in every way i can .

phoddie, 2022-07-15:
@spillz-dxb – let's start with an example of the Function node using an external standard ECMAScript module from the Moddable SDK. For this example, let's use the [Preference module](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/files/files.md#preference) to store state across runs.

Here's an example flow to do that.

<details>
<summary>Function node using Preference module</summary>

```json
[
    {
        "id": "50be8a0703e59b54",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "07ec664a0c91cd4a",
        "type": "inject",
        "z": "50be8a0703e59b54",
        "name": "",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": true,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 90,
        "y": 200,
        "wires": [
            [
                "263164d2ce076cf0"
            ]
        ]
    },
    {
        "id": "263164d2ce076cf0",
        "type": "function",
        "z": "50be8a0703e59b54",
        "name": "",
        "func": "const nowSeconds = Math.round(parseInt(msg.payload) / 1000);\nconst lastRun = Preference.get(\"node-red-mcu\", \"lastRun\");\nPreference.set(\"node-red-mcu\", \"lastRun\", nowSeconds);\nif (undefined === lastRun)\n    msg.payload = \"first time\";\nelse {\n    const elapsedSeconds = nowSeconds - lastRun;\n    msg.payload = `last run ${elapsedSeconds} seconds ago`;\n}\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [
            {
                "var": "Preference",
                "module": "preference"
            }
        ],
        "x": 300,
        "y": 200,
        "wires": [
            [
                "a6a95537d268828b"
            ]
        ]
    },
    {
        "id": "a6a95537d268828b",
        "type": "debug",
        "z": "50be8a0703e59b54",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": true,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 520,
        "y": 200,
        "wires": []
    }
]
```
</details>

To run that, you'll need to add the Preferences module to the `"includes"` section of node-red-mcu/manifest.json.

```js
{
	"include": [
		...
		"$(MODDABLE)/modules/files/preference/manifest.json"
	]
	...
```

This technique works with any module in the Moddable SDK. Maybe give that a try to see how it works for you. Once that works, we can look at how to make some simple changes to the CommonJS modules used by the fulll Node-RED to convert them to standard ECMAScript modules so they can be imported too.

spillz-dxb, 2022-07-16:
> This technique works with any module in the Moddable SDK. Maybe give that a try to see how it works for you. Once that works, we can look at how to make some simple changes to the CommonJS modules used by the fulll Node-RED to convert them to standard ECMAScript modules so they can be imported too.

[image]

Done!

phoddie, 2022-08-09:
@spillz-dxb – Apologies for the delay. I've been working on a big revision to Node-RED MCU. Last night I pushed that to this repository. It has many improvements. One is that it is able to run more CommonJS modules directly. The project now includes the Delay node from the full Node-RED as an example of that. If you compare [the version](https://github.com/phoddie/node-red-mcu/blob/beaffa710f3623a9be7ce93acb62ef744f3c8f78/nodes/delay/89-delay.js#L17-L22) in the Node-RED MCU repository with the original, you'll see just a few changes to reduce memory use.

The Delay node is doesn't have any dependencies on Node.js. It is pure JavaScript plus the use of setInterval for timers. I'm not sure how many other nodes are like this. I think the Delay node might work and that it could be possible to use the full Filter node in place of my lightweight version. Maybe you can try some others from the Node-RED core or perhaps some contributed nodes? Maybe that will identify some additional work to do to make more Node-RED MCU Edition able to run more existing nodes.

phoddie, 2022-09-05:
> Next is trying out servers and listeners !

Coming back to this comment, `HTTP In`,`HTTP Response`, `UDP In` and `UDP Out` are all now available. It would be great if you can try them out. The readme describes what is (and isn't yet implemented) for the [HTTP](https://github.com/phoddie/node-red-mcu/blob/main/readme.md#http-in) and [UDP](https://github.com/phoddie/node-red-mcu/blob/main/readme.md#udp-in) nodes.

phoddie, 2022-10-19:
Closing this out due to inactivity.

closed 2022-10-19 by phoddie