phoddie/node-red-mcu #18: nodered2mcu: Need to escape quotation marks in node name

ralphwetzel, 2022-08-26:
Hi!
nodered2mcu creates defective code as it doesn't escape `"` in node names.

Test case:
```
[
    {
        "id": "2a0f075a689436c6",
        "type": "tab",
        "label": "Escape Test",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": true
    },
    {
        "id": "9957cd4fd3bb3a7b",
        "type": "debug",
        "z": "2a0f075a689436c6",
        "name": "\"Test\"",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": true,
        "x": 330,
        "y": 160,
        "wires": []
    }
]
```

Part of the `flows.js`, as generated by nodered2mcu:

```
// auto-generated by nodered2mcu on Fri Aug 26 2022 07:32:03 GMT+0200

import "nodered";		// ensure globalThis.RED is available
import Timer from "timer"

function build(flows, createFlow, createNode) {
	let flow, node, nodes;
	flow = createFlow("__config", "global config")
	flow = createFlow("2a0f075a689436c6", "")
	createNode("debug", "9957cd4fd3bb3a7b", ""Test"", flow);            <<==

	flows = flows.values();
	flow = flows.next().value;	// __config
```

Should be easy to fix...

phoddie, 2022-08-26:
@ralphwetzel – good catch. Thank you. I've update `nodered2mcu` to string escape both node and flow names.

And your timing is good. I have several other fixes pending related to string escapes and property names. They should all be available later today.

phoddie, 2022-08-26:
The update to `nodered2mcu` is live. Just grab the latest Moddable SDK, rebuild tools, and you should be good.

closed 2022-10-10 by phoddie