phoddie/node-red-mcu #23: Exception: LinkOutNode.prototype.onMessage: cannot coerce undefined to object!

ralphwetzel, 2022-10-09:
The design of the (standard) LinkNodes looks a bit strange to me: Although the `LinkOutNode` shown in the picture definitely has no link to another (`LinkIn`) node, it's `links` property is populated. The id shown does not exist (anymore?) in any of my flows.

[image]

In `nodered.js` the call @ line 849 to `RED.nodes.getNode(link)` w/ this non-existing id returns `undefined`: 
https://github.com/phoddie/node-red-mcu/blob/7d9f7630e0bb17d00edd00ab56f32758acc66789/nodered.js#L842-L850

Later, when a msg arrives, `links[i]` @ `links[i].send(msg)` (line 855) will be that returned `undefined`. Consequentially, an exception is thrown: `LinkOutNode.prototype.onMessage: cannot coerce undefined to object!`
https://github.com/phoddie/node-red-mcu/blob/7d9f7630e0bb17d00edd00ab56f32758acc66789/nodered.js#L851-L861

I propose to replace line 849 by a classical loop checking if the (link) id references an existing node:

``` javascript
 			this.#links = [];
 			for (let i=0, l=config.links.length; i<l; i++) {
 				let n = RED.nodes.getNode(config.links[i]);
 				if (n) this.#links.push(n);
 			}
```

phoddie, 2022-10-09:
If I understand the report correctly, the problem is that the link is broken -- that one or more of the node ID values in the `links` array references a node that no longer exists? I had expected Node-RED would clean that up. If that is really happening, then that's something that `nodered2mcu` should clean-up so the runtime doesn't have to be account for it. Would you mind posting an example `flows.json` that shows the problem so I can investigate? Thank you!

ralphwetzel, 2022-10-09:
Here's a flow showing the same issue.

[image]

It even references two non existing nodes in its `links` property:

``` json
[
    {
        "id": "f36e3774a30673e3",
        "type": "tab",
        "label": "Flow 6",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": false
        }
    },
    {
        "id": "b1ee8cbb775e3213",
        "type": "link in",
        "z": "f36e3774a30673e3",
        "name": "link in 36",
        "links": [
            "7ecd205bebc32369"
        ],
        "_mcu": {
            "mcu": false
        },
        "x": 555,
        "y": 540,
        "wires": [
            [
                "1ecc56b11d08ed72"
            ]
        ]
    },
    {
        "id": "7ecd205bebc32369",
        "type": "link out",
        "z": "f36e3774a30673e3",
        "name": "link out 46",
        "mode": "link",
        "links": [
            "316aa1d3398a969d",
            "b2d692753dcb51e4",
            "b1ee8cbb775e3213"
        ],
        "_mcu": {
            "mcu": false
        },
        "x": 485,
        "y": 540,
        "wires": []
    },
    {
        "id": "1ecc56b11d08ed72",
        "type": "debug",
        "z": "f36e3774a30673e3",
        "name": "debug 32",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": false
        },
        "x": 660,
        "y": 540,
        "wires": []
    },
    {
        "id": "cf0c5a63e767b787",
        "type": "inject",
        "z": "f36e3774a30673e3",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "_mcu": {
            "mcu": false
        },
        "x": 380,
        "y": 540,
        "wires": [
            [
                "7ecd205bebc32369"
            ]
        ]
    }
]
```

phoddie, 2022-10-09:
Thank you for the flow. I have an update to `nodered2mcu` that removes broken references in the `links` array. That avoids the runtime error later. That'll be included in a Moddable SDK update in a day or so.

I also fixed some bit-rot that caused the `link call` nodes to fail. The problem was introduced when the switch was made to use `structuredClone` to duplicate messages. That's committed to the Node-RED MCU repository.

phoddie, 2022-10-18:
I think this is now fixed. Would you please confirm?

ralphwetzel, 2022-10-18:
It is! Thank you! 👍

closed 2022-10-18 by ralphwetzel