phoddie/node-red-mcu #43: LinkOut node doesn't return to LinkCall Node

ralphwetzel, 2022-11-21:
@phoddie --
As the title says: A LinkOut node doesn't return to its LinkCall Node.

<details>
  <summary>Test case</summary>

[image]

```js
[
    {
        "id": "5d96ad23f6c6c847",
        "type": "tab",
        "label": "Flow 15",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": true
        }
    },
    {
        "id": "cc2784ee66765e42",
        "type": "group",
        "z": "5d96ad23f6c6c847",
        "name": "Come here!",
        "style": {
            "label": true
        },
        "nodes": [
            "988cf8d2c744c76e",
            "4ea66c21ba624d51",
            "300e0c35ed5e6039"
        ],
        "x": 254,
        "y": 419,
        "w": 362,
        "h": 82
    },
    {
        "id": "6ed2343e18f0a785",
        "type": "inject",
        "z": "5d96ad23f6c6c847",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "_mcu": {
            "mcu": true
        },
        "x": 340,
        "y": 380,
        "wires": [
            [
                "1613d9f16b6133d2"
            ]
        ]
    },
    {
        "id": "1613d9f16b6133d2",
        "type": "link call",
        "z": "5d96ad23f6c6c847",
        "name": "",
        "links": [
            "988cf8d2c744c76e"
        ],
        "linkType": "static",
        "timeout": "30",
        "_mcu": {
            "mcu": true
        },
        "x": 510,
        "y": 380,
        "wires": [
            [
                "1a9d7ab025d3035c"
            ]
        ]
    },
    {
        "id": "988cf8d2c744c76e",
        "type": "link in",
        "z": "5d96ad23f6c6c847",
        "g": "cc2784ee66765e42",
        "name": "Come here!",
        "links": [],
        "_mcu": {
            "mcu": true
        },
        "x": 295,
        "y": 460,
        "wires": [
            [
                "4ea66c21ba624d51"
            ]
        ]
    },
    {
        "id": "4ea66c21ba624d51",
        "type": "change",
        "z": "5d96ad23f6c6c847",
        "g": "cc2784ee66765e42",
        "name": "payload > done",
        "rules": [
            {
                "t": "set",
                "p": "payload",
                "pt": "msg",
                "to": "done",
                "tot": "str"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "_mcu": {
            "mcu": true
        },
        "x": 430,
        "y": 460,
        "wires": [
            [
                "300e0c35ed5e6039"
            ]
        ]
    },
    {
        "id": "300e0c35ed5e6039",
        "type": "link out",
        "z": "5d96ad23f6c6c847",
        "g": "cc2784ee66765e42",
        "name": "link out 82",
        "mode": "return",
        "links": [],
        "_mcu": {
            "mcu": true
        },
        "x": 575,
        "y": 460,
        "wires": []
    },
    {
        "id": "1a9d7ab025d3035c",
        "type": "debug",
        "z": "5d96ad23f6c6c847",
        "name": "debug 55",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": true
        },
        "x": 680,
        "y": 380,
        "wires": []
    }
]
```

</details>

https://github.com/phoddie/node-red-mcu/blob/6427d0e16c32c9d32d29d117ff5c59a79d6da51c/nodered.js#L951-L980

This is due to the fact, that `if (links) {` is `true` in line 961 for `links === []` and `nodered2mcu.js` emits `links: [],` for this node.

```js
	[...]
	node = nodes.next().value;	// link out - 300e0c35ed5e6039
	node.onStart({
		g: "cc2784ee66765e42",
		name: "link out 82",
		mode: "return",
		links: [],
	});
	[...]
```
I propose to switch the sequence of the `if` clauses - as the existence of `msg._linkSource` should be the trigger to return to the calling node.

```js
	if (Array.isArray(msg._linkSource) && (msg._linkSource.length > 0)) { [...] }
	else if (links) { [...] }
	else { [...] }
```

phoddie, 2022-11-21:
@ralphwetzel, thank you for the test case and analysis! The tests I created work, so something must be different. I'll take look.

ralphwetzel, 2022-11-21:
There seems to be another issue related to the _LinkCall_ topic:

The `timeout` value is - by reference to the UI - defined in `seconds`:

[image]

I'm yet pretty sure, `Timer.set()` takes the timeout value in `ms`. 

https://github.com/phoddie/node-red-mcu/blob/6427d0e16c32c9d32d29d117ff5c59a79d6da51c/nodered.js#L923-L930

This currently creates numerous `link error missingReturn` warnings - as the timeout fires before the function call returns...

phoddie, 2022-11-21:
Could you  be using an out-of-date `nodered2mcu`? The [preprocessing](https://github.com/Moddable-OpenSource/moddable/blob/76ae9d8a5f456078dc9f06a695b17b2a621a8208/tools/nodered2mcu.js#L900
) does the conversion from seconds to milliseconds.

phoddie, 2022-11-22:
Interesting example. I never thought about putting a Link In node in the same flow as the Link Call node. I just tried this with the last MCU Edition (including the current `nodered2mcu` from the Moddable SDK) and it outputs a debug message with "done" as expected. I'm hoping that after you update, it will work for you as well.

ralphwetzel, 2022-11-22:
> Could you be using an out-of-date `nodered2mcu`?

Indeed. I'm pretty sure though that GitHub showed me (for the last several days) [this](https://github.com/Moddable-OpenSource/moddable/commit/76ae9d8a5f456078dc9f06a695b17b2a621a8208) commit as the current one - and now magically jumped forward. 😕 Anyway. I'll test with the latest version...

ralphwetzel, 2022-11-22:
👍 Both observations are resolved w/ the current commits. Thank you!

closed 2022-11-22 by ralphwetzel

phoddie, 2022-11-22:
Glad it was that easy. Sorry for the headache. I did learn that Link nodes can link into their own flow from this.