phoddie/node-red-mcu #44: MQTT topics

ralphwetzel, 2022-11-24:
Peter --

I tried to use the MQTT nodes - and encountered some strange effects:

MQTT Sender (to be run on the MCU):
[image]

<details>
  <summary>sender.json</summary>

```js
[
    {
        "id": "4dc6e91342cfdc95",
        "type": "tab",
        "label": "MQTT Sender",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": true
        }
    },
    {
        "id": "7e8450f889bc77e5",
        "type": "group",
        "z": "4dc6e91342cfdc95",
        "name": "Test: Send to MQTT",
        "style": {
            "label": true
        },
        "nodes": [
            "329f102a0e2672e9",
            "f7d2c588db6cf95c",
            "23d2e1e9ec0bf70e"
        ],
        "x": 494,
        "y": 139,
        "w": 552,
        "h": 82
    },
    {
        "id": "329f102a0e2672e9",
        "type": "function",
        "z": "4dc6e91342cfdc95",
        "g": "7e8450f889bc77e5",
        "name": "prepare for MQTT",
        "func": "// msg.topic == sensor.id\n\nlet values = {\n    \"t1\": msg.data.t1,\n    \"t2\": msg.data.t2,\n    \"t3\": msg.payload\n}\n\nlet res = [];\n\nfor (let key in values) {\n    if (values[key]) {\n        let m = {\n            \"topic\": `mqtt/test/${key}`,\n            \"payload\": values[key]\n        }\n        res.push(m);\n    }\n}\n\n// send a msg per value\nreturn [res];\n",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "_mcu": {
            "mcu": true
        },
        "x": 790,
        "y": 180,
        "wires": [
            [
                "f7d2c588db6cf95c",
                "7d8c6612a681c9cb"
            ]
        ],
        "inputLabels": [
            "Sensor values"
        ],
        "outputLabels": [
            "MQTT messages"
        ]
    },
    {
        "id": "f7d2c588db6cf95c",
        "type": "mqtt out",
        "z": "4dc6e91342cfdc95",
        "g": "7e8450f889bc77e5",
        "name": "",
        "topic": "",
        "qos": "2",
        "retain": "false",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "4b0b89cd78b32db9",
        "_mcu": {
            "mcu": true
        },
        "x": 970,
        "y": 180,
        "wires": []
    },
    {
        "id": "23d2e1e9ec0bf70e",
        "type": "inject",
        "z": "4dc6e91342cfdc95",
        "g": "7e8450f889bc77e5",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "data",
                "v": "{}",
                "vt": "json"
            },
            {
                "p": "data.t1",
                "v": "2",
                "vt": "str"
            },
            {
                "p": "data.t2",
                "v": "3",
                "vt": "str"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "01",
        "payload": "1",
        "payloadType": "num",
        "_mcu": {
            "mcu": true
        },
        "x": 590,
        "y": 180,
        "wires": [
            [
                "329f102a0e2672e9"
            ]
        ]
    },
    {
        "id": "7d8c6612a681c9cb",
        "type": "debug",
        "z": "4dc6e91342cfdc95",
        "name": "debug 67",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": true
        },
        "x": 1020,
        "y": 300,
        "wires": []
    },
    {
        "id": "4b0b89cd78b32db9",
        "type": "mqtt-broker",
        "name": "",
        "broker": "192.168.178.72",
        "port": "1883",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": "",
        "_mcu": {
            "mcu": false
        }
    }
]
```

</details>

MQTT Receiver (to be run as standard NR flow):
[image]

<details>
  <summary>receiver.json</summary>

```js
[
    {
        "id": "df3521992a1ba688",
        "type": "tab",
        "label": "MQTT Receiver",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": false
        }
    },
    {
        "id": "e2e71ed681ee60da",
        "type": "mqtt in",
        "z": "df3521992a1ba688",
        "name": "",
        "topic": "mqtt/test/#",
        "qos": "2",
        "datatype": "auto-detect",
        "broker": "4b0b89cd78b32db9",
        "nl": false,
        "rap": true,
        "rh": 0,
        "inputs": 0,
        "_mcu": {
            "mcu": false
        },
        "x": 291.6666564941406,
        "y": 224.99998474121094,
        "wires": [
            [
                "1f1bcb4d8b228924"
            ]
        ]
    },
    {
        "id": "1f1bcb4d8b228924",
        "type": "debug",
        "z": "df3521992a1ba688",
        "name": "debug 68",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": false
        },
        "x": 521.6666564941406,
        "y": 224.99998474121094,
        "wires": []
    },
    {
        "id": "4b0b89cd78b32db9",
        "type": "mqtt-broker",
        "name": "",
        "broker": "192.168.178.72",
        "port": "1883",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": "",
        "_mcu": {
            "mcu": false
        }
    }
]
```

</details>

👉 You need to configure the MQTT connection to adapt to your system!

Observations:
* When pushing the inject button for the first time, the sender emits the Debug messages, yet no data is received by the receiver.
* When pushing the inject button for the second time, the sender throws with `/Users/ralph/Projekte/Moddable/moddable/examples/io/tcp/mqttclient/mqttclient.js (121) # Exception: MQTTClient.prototype.write: cannot coerce undefined to object!`.

When both flows are run as standard NR flows, everything works as intended.
Any proposals?

My system status: Moddable @ [758b624](https://github.com/Moddable-OpenSource/moddable/commit/758b624dc985362c77cd283739bec7f87bc28ac2), node-red-mcu @ [6427d0e](https://github.com/phoddie/node-red-mcu/commit/6427d0e16c32c9d32d29d117ff5c59a79d6da51c)

phoddie, 2022-11-25:
Thanks for the report. The problem appears to have that the sender was not handling the topic correctly in the configuration JSON. I've committed a fix and tested successfully with the test.mosquitto.org broker. `nodered2mcu` should be updated to normalize this, so the runtime doesn't have to.

ralphwetzel, 2022-11-25:
Just tested this fix successfully. :+1:
I propose to leave the report open until you put it into `nodered2mcu`.

phoddie, 2022-11-25:
Thanks for the quick check! It makes sense to leave this open as a reminder to update `nodered2mcu`. I've done that just now and the changes will propagate out next week. As a nice side-effect, I found that the four private fields on the "mqtt in" node can be eliminated, which saves a bit of memory for each instance.

phoddie, 2022-12-02:
The `nodered2mcu` updates are live, so I've committed the corresponding changes to the MQTT node implementation. I think we can close this out now?

ralphwetzel, 2022-12-02:
Confirmed. 👍 Yes we can...

closed 2022-12-02 by ralphwetzel