phoddie/node-red-mcu #56: Driving Moddable Two screen backlight GPIO PWM 18 on Node-RED MCU

rei-vilo, 2022-12-14:
One more question related to GPIOs, this time `PWM 18` in charge of [backlight](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/moddable-two.md#backlight) from Node-RED MCU.

+ Configuration the RPI-GPIO node with `PWM 18` shows `12 not available`. 
[image]

+ Trying to access `config.brightness = 75;" or `backlight.write(80);` return errors as well.
[image]

+ The panes **Configuration nodes** and **Context** do not list this parameter or are empty.

How to manage backlight with Node-RED?

_Originally posted by @rei-vilo in https://github.com/phoddie/node-red-mcu/discussions/49#discussioncomment-4402473_

phoddie, 2022-12-16:
@rei-vilo – I'm not able to reproduce the problem using a Moddable Two. Here's a flow that changes the backlight to a random value once a second.

[image]

<details>
<summary>Flow</summary>

```json
[
    {
        "id": "bc3f7d918132436a",
        "type": "tab",
        "label": "delete me",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": false
        }
    },
    {
        "id": "71c961cee3ed80f6",
        "type": "rpi-gpio out",
        "z": "bc3f7d918132436a",
        "name": "backlight",
        "pin": "18",
        "set": "",
        "level": "0",
        "freq": "",
        "out": "pwm",
        "bcm": true,
        "_mcu": {
            "mcu": false
        },
        "x": 560,
        "y": 900,
        "wires": []
    },
    {
        "id": "7290773e82e7274c",
        "type": "inject",
        "z": "bc3f7d918132436a",
        "name": "",
        "props": [],
        "repeat": "1",
        "crontab": "",
        "once": true,
        "onceDelay": 0.1,
        "topic": "",
        "_mcu": {
            "mcu": false
        },
        "x": 190,
        "y": 900,
        "wires": [
            [
                "f214cc8f4e11160c"
            ]
        ]
    },
    {
        "id": "f214cc8f4e11160c",
        "type": "random",
        "z": "bc3f7d918132436a",
        "name": "",
        "low": "0",
        "high": "100",
        "inte": "true",
        "property": "payload",
        "_mcu": {
            "mcu": false
        },
        "x": 360,
        "y": 900,
        "wires": [
            [
                "71c961cee3ed80f6"
            ]
        ]
    }
]
```

</details>

I run that by updating the flows.json manually and using `mcconfig -d -m -p esp32/moddable_two`.

Maybe you are doing something different here?

rei-vilo, 2022-12-18:
Thank you for your answer. 

As the symptoms were not new, I cleaned the `build/tmp` and `build/bin` sub-folders.

Everything works as expected now, except PWM seems to be reversed: 100 is dark and 0 is bright.

closed 2022-12-18 by rei-vilo

phoddie, 2022-12-18:
> Everything works as expected now...

Great!

> ... except PWM seems to be reversed: 100 is dark and 0 is bright

Correct. The PWM on the Moddable Two backlight interprets 0 as full brightness and 100 as off. The `Backlight` in the host [takes care of that automatically](https://github.com/Moddable-OpenSource/moddable/blob/7be4b2343b56f9341bf48be36ec2b4edbf6e9e8e/build/devices/esp32/targets/moddable_two/setup-target.js#L13). But here, since you are writing to the PWM directly, you need to apply the mapping. In case you aren't aware, the Node-RED Range node can do that mapping:

[image]