phoddie/node-red-mcu #79: Question on Dashboard

sleveel, 2023-01-30:
Hi,

As I understand it, no dashboard nodes are supported.
What would you recommend to create something similar to a Node-Red dashboard on a Moddable Two?

Many thanks,

phoddie, 2023-01-30:
> ...no dashboard nodes are supported...

UI nodes are supported, as @ralphwetzel notes. I demonstrated them at Node-RED Con 2022. Check out the [video](https://www.youtube.com/watch?v=2T5mraNohGc).

sleveel, 2023-01-30:
Of course...
I had seen the video.
Sorry... it slipped my mind.

On my way to test it.
Thanks!

sleveel, 2023-02-01:
One last question, if the flow and its associated dashboard is deployed on a Moddable Two, is there a way to display the same on a mobile phone if it is on the same network?

Should I deploy the flow on the phone with a Java emulator or is there a way to mirror the screen and associated touch inputs?

phoddie, 2023-02-01:
The microcontroller on Moddable Two is working hard running Node-RED, especially with the dashboard. It can't help with the mobile UI. It can share its data using the MQTT or the HTTP nodes. For the phone UI, you'd need to deploy a second version of the project to the phone, but that's not something I know how to do.

phoddie, 2023-02-13:
This looks to be resolved, so I'm closing it out.

closed 2023-02-13 by phoddie