phoddie/node-red-mcu #87: ESP8266 with MQTT, Trigger and GPIO node not functioning

colinl, 2023-02-17:
This simple flow, with MQTT passthrough from MQTT In to MQTT Out, and an Inject and Trigger node driving a Wemos D1 Mini onboard LED is not functioning correctly.

In xsbug I see the wifi connecting, but the LED is not flashing and when I publish to the MQTT In topic, noting happens.
A breakpoint on nodes/trigger/89-trigger.js line 32 (`RED.nodes.createNode(this,n);`) is never hit.

The Instruments tab in xsbug shows it running something, but there is little happening after the first few seconds.

To run it I use the command 
`mcconfig -d -m -p esp/nodemcu ssid="**" password="**"`

If I remove the MQTT nodes the LED flashes correctly.

Do I have to do something to enable MQTT?

```
[{"id":"ed0d63aad317e829","type":"inject","z":"f92bb64b93bfc46e","name":"","props":[{"p":"payload"}],"repeat":"2","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"1","payloadType":"num","_mcu":{"mcu":true},"x":170,"y":400,"wires":[["a83bf3122efb505c"]]},{"id":"a83bf3122efb505c","type":"trigger","z":"f92bb64b93bfc46e","name":"","op1":"0","op2":"1","op1type":"num","op2type":"num","duration":"500","extend":false,"overrideDelay":false,"units":"ms","reset":"","bytopic":"all","topic":"topic","outputs":1,"_mcu":{"mcu":true},"x":360,"y":400,"wires":[["32dcd5abbb576d04"]]},{"id":"32dcd5abbb576d04","type":"rpi-gpio out","z":"f92bb64b93bfc46e","name":"","pin":"2","set":"","level":"0","freq":"","out":"out","bcm":true,"_mcu":{"mcu":true},"x":530,"y":400,"wires":[]},{"id":"90e9ccdae3d60741","type":"mqtt out","z":"f92bb64b93bfc46e","name":"","topic":"test/mcu/pid/result","qos":"","retain":"","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"75fb29423f8a0770","_mcu":{"mcu":true},"x":610,"y":140,"wires":[]},{"id":"0b4b04cb671e3bca","type":"debug","z":"f92bb64b93bfc46e","name":"debug 92","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","_mcu":{"mcu":true},"x":580,"y":80,"wires":[]},{"id":"8c687e519a7bd590","type":"mqtt in","z":"f92bb64b93bfc46e","name":"","topic":"test/mcu/pid/set","qos":"1","datatype":"json","broker":"75fb29423f8a0770","nl":false,"rap":true,"rh":0,"inputs":0,"_mcu":{"mcu":true},"x":100,"y":80,"wires":[["90e9ccdae3d60741","0b4b04cb671e3bca"]]},{"id":"75fb29423f8a0770","type":"mqtt-broker","name":"Owl2 for mcu","broker":"192.168.49.83","port":"1883","clientid":"","autoConnect":true,"usetls":false,"protocolVersion":"4","keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","birthMsg":{},"closeTopic":"","closeQos":"0","closePayload":"","closeMsg":{},"willTopic":"","willQos":"0","willPayload":"","willMsg":{},"userProps":"","sessionExpiry":"","_mcu":{"mcu":false}}]
```

phoddie, 2023-02-17:
I ran your flow here on an ESP8266. The LED flashes and, after updating the MQTT broker to `test.mosquitto.org`, it relays messages received on `test/mcu/pid/set` to `test/mcu/pid/result` (confirmed using mosquitto_pub/sub).

Since both MQTT and Trigger are not working, it sounds like your flow just isn't running. A good way to check that is to look at the generated flows.js in the debugger. Since you are building for esp/nodemcu, that should be located at $MODDABLE/build/tmp/esp/nodemcu/debug/node-red-mcu/modules/flows.js.  There you will see a `build` function. Set a breakpoint there and you can step through the instantiation of each node. That's a good way to confirm that the expected nodes are present and that they have the expected configuration. You can also do things like break when the Inject node's timer triggers, to confirm that it is injecting messages as expected.

```js
		initialize: function () {
			Timer.set(() => this.trigger(), 2000, 2000);
		},
```

colinl, 2023-02-17:
Thanks for coming back so quickly.

I see the problem, when I looked in that file I saw that it was instantiating only the broker node, and when I compared it with yours I saw that yours has entries for the flow tabs.  I was exporting only the nodes rather than a complete tab. It does say in the docs to export flows, but I wonder whether a note in large letters warning that complete tabs must be exported would be a good idea.  Then those of us that, foolishly, only skim the docs would be less likely to get it wrong. :)
All working perfectly now, sorry about the noise.

closed 2023-02-17 by colinl

phoddie, 2023-02-17:
> Then those of us that, foolishly, only skim the docs would be less likely to get it wrong. :)

Yes, well. There's a related challenge with the UI nodes, where it is necessary to report **all flows** to have the dashboard configuration node included.

I'll update the docs with a bit more text. Maybe I can also update `nodered2mcu` to issue a warning if there's a potentially missing node (e.g. no broker node for an MQTT Out). That could give a hint.

> All working perfectly now, sorry about the noise.

No problem. I appreciate the clear report and your patience in working through issues.