phoddie/node-red-mcu #92: Organisation for multiple projects?

colinl, 2023-02-20:
If I want to have more than one node-red-mcu project in development at the same time is there a recommended way of organising things?  Ideally I would like the project folder separated from the node-red-mcu folder if possible. At the moment it appears the node-red-mcu must be a child of the project folder, unless I am missing something.

Sineos, 2023-02-20:
When playing with this, I use the NR command line arguments to start with different flow files, e.g.:
`node-red --settings settings_1.js flows_1.json`

This way you can even have different 
```
process.env.MODDABLE = "C:/Users/souls/Projects/moddable"
process.env.IDF_PATH = "C:/Espressif/frameworks/esp-idf-v4.4.3"
```
if needed

colinl, 2023-02-20:
I have been doing exactly that, but how do I then build the one I want to build without copying it across to Projects?

phoddie, 2023-02-20:
@colinl – I don't have a clear enough picture of how the pieces fit together to offer a suggestion yet. In fact, I've always found switching "projects" in Node-RED itself to be awkward (either export / delete / import in the editor, or stop Node-RED/move flows file/restart). Maybe you can share how you are managing that? And from there, I might understand the file organization well enough to offer some suggestion.

Sineos, 2023-02-20:
The documentation says you can also pass a project name instead of the flow file. Never tried.
My workflow is simple: Have one default installation and simply start a new flows.json when I want to do something new with the MCU

colinl, 2023-02-21:
I don't use node-red projects, I was using the word generically.  In fact I use a different .node-red folder for each 'project' and each one has its own git repository for source control.  So for my wemos D1 Mini devices I might have ~/.node-red_we01, .node-red_we02, etc.  I then want to build the flow file there using node-red-mcu.  Perhaps the best thing is to install node-red-mcu as a git subproject under each node-red folder so I have control over which version of that is used.   On second thoughts that doesn't quite work as the flows file needs to be inside the node-red-mcu folder.

phoddie, 2023-02-21:
@colinl, thanks for the details. 

> that doesn't quite work as the flows file needs to be inside the node-red-mcu folder

The Moddable SDK manifest gives quite a bit of flexibility here. One option is to put a copy of the [manifest.json](https://github.com/phoddie/node-red-mcu/blob/main/manifest.json) in your project directory (~/.node-red_we01/manifest.json). You will need to update the path to manifest_host.json to point to your Node-RED MCU git repository. Then you can just `cd ~/.node-red_we01/manifest.json & mcconfig...` to build.

colinl, 2023-02-21:
That could work. Thanks. I will give it a go.

colinl, 2023-02-22:
OK, this is working well for me:
1. The .node-red folder is under git control (or whichever folder contains the node-red flows file)
2. Add node-red-mcu as a git submodule under .node-red
3. Copy node-red-mcu/manifest.json into .node-red and `git add` it
4. Edit manifest.json to allow building from the .node-red folder. For example
```
{
	"include": [
		"./node-red-mcu/manifest_host.json"
	],
	"modules": {
		"*": [
			"./node-red-mcu/main",
			{
				"source": "./flows",
				"transform": "nodered2mcu"
			}
		]
	},
	"preload": [
		"flows"
	]
}
```
5. If you have any additional installed nodes then create a folder `contrib-nodes` and put the manifest files for those in that folder.  For example, I use node-red-contrib-pid and node-red-contrib-timeprop so I create files manifest-node-red-contrib-pid.manifest and similar for timeprop.  In the pid manifest is this, which includes the contrib node source from the node_modules folder
```
{
	"modules": {
		"*": "../node_modules/node-red-contrib-pid/pid"
	},
	"preload": "pid"
}
```
6. Git add those manifest files
7. Add them into the `include` section of the main manifest file, so mine now reads
```
	"include": [
		"./node-red-mcu/manifest_host.json",
		"./contrib-nodes/manifest-node-red-contrib-timeprop.json",
		"./contrib-nodes/manifest-node-red-contrib-pid.json"
	],
```
Now everything is controlled by git, and the project can be built from the .node-red folder using something like
`mcconfig -d -m -p esp/nodemcu`

phoddie, 2023-02-22:
Glad that works. What you describe all makes good sense. Thanks for sharing the details. That will be helpful for others.

closed 2023-02-22 by phoddie