phoddie/node-red-mcu #98: How does one use the MCU Analog node?

colinl, 2023-03-05:
In the readme it describes the MCU Analog input node, but it is not clear to me how to configure that.

phoddie, 2023-03-05:
> but it is not clear to me how to configure that

I'm not 100% certain what the question is. Are you asking how to install the node into the Node-RED editor? If so, it is installed as part of the MCU Sensor node, which contains a bundle of I/O related nodes.

[image] 
[image]

colinl, 2023-03-06:
Yes, that is the question that I failed to put clearly.  I did not realise that it is included in the Sensor node. The fact that it doesn't appear in the list of supported types in https://github.com/phoddie/node-red-mcu/blob/main/nodes/sensor/readme.md misled me.  I will look at submitting a PR for the documentation, after checking that all the the others are covered.

Analogue In appears to be working well on the D1 Mini.

Thanks again.

phoddie, 2023-03-06:
> Analogue In appears to be working well on the D1 Mini

Perfect.

Regarding the documentation, it is a little tricky to explain well. The sensor node directory has grown beyond its original scope to include all sorts of related things. The `reorg` branch of this repository restructures things a bit to help manage that. We hope to merge that soon. Help in documenting that better, through suggestions or PRs, would be most welcome.

colinl, 2023-03-06:
Should I wait till the reorg is done then?

phoddie, 2023-03-06:
That's a good idea. It should just be a day or so.

phoddie, 2023-03-09:
The re-org branch has been merged.

phoddie, 2023-04-10:
I think we can close this out?

colinl, 2023-04-10:
I keep meaning to look again at the readme to see if I can improve it at all, but haven't got round to it with pressure of other stuff.

phoddie, 2023-04-10:
No problem. We can leave this open as a reminder. Good to have the note that this is down to documentation.

colinl, 2023-04-12:
In fact, since the mcu package will hopefully be released via npm some time reasonably soon perhaps there is not much point adjusting the docs now, as they will have to be adjusted then anyway.

phoddie, 2023-04-12:
Fair point. FWIW – I think I'm on course to be able to try publishing those nodes to npm this weekend. The work on MCU Restart and OTA Update nodes have helped me understand how all the pieces fit together.

phoddie, 2023-04-17:
First pass at getting the MCU IO nodes into npmjs is up. It is working well for me, at least. ;) There's work to do on docs before adding to the Node-RED Library. 

FWIW – one nice side effect of these changes is that is a step to generate firmware that only contains the nodes used by the project. That's been a goal of mine for some time. Smaller firmware images will download more quickly and leave more free space for flows and their nodes. That's going to be really nice on devices with the most limited space, like ESP8266. There's some more work to get there, but it shouldn't be too far off.