phoddie/node-red-mcu #103: MQTT-Out node does not work because of missing broker config node

bartbutenaers, 2023-03-16:
Hi @phoddie,

I'm afraid I need your assistance again.

I had added an MQTT-Out node to my flow, but after deploying I saw this error in the xterm terminal of Ralph's sidebar:
```
>> IDF_PYTHON_ENV_PATH: /home/pi/.espressif/python_env/idf4.4_py3.9_env
>> mcconfig -d -x localhost:5004 -l -v -m -p esp32/wt32_eth01 -t xsbug ssid=xxx password=yyy
/home/pi/esp-idf/tools/idf.py
# starting xsbug
PORT_USED=$(grep 'Serial port' /home/pi/Projects/moddable/build/tmp/esp32/wt32_eth01/debug/t5rg16s07b/xsProj-esp32/flashOutput | awk 'END{print($3)}'); \
cd /home/pi/Projects/moddable/tools/xsbug-log && node xsbug-log bash -c \"XSBUG_PORT=5004 XSBUG_HOST=localhost serial2xsbug /dev/ttyUSB0 460800 8N1\"
# xsbug-log listening on port 5002. ^C to exit.
# Connected to "main"
Skipping Ethernet setup because Wi-Fi SSID is configured.
Wi-Fi connected to "xxx"
IP address xxxxxxx
# Break: MQTTOutNode.prototype.onStart: cannot coerce undefined to object!
  #0: MQTTOutNode.prototype.onStart /home/pi/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/node-red-mcu/nodered.js:1247
  #1: build /home/pi/Projects/moddable/build/tmp/esp32/wt32_eth01/debug/t5rg16s07b/modules/flows.js:59
  #2: RED.build /home/pi/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/node-red-mcu/nodered.js:163
  #3: main /home/pi/.node-red/mcu-plugin-cache/t5rg16s07b/main.js:3
  #4: (host)
  #5: @anonymous-2193
  #6: (anonymous-649) /home/pi/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/node-red-mcu/setupwifi.js:47
  #7: (anonymous-812) /home/pi/Projects/moddable/examples/network/wifi/wificonnection/wificonnection.js:95
  #8: (host)
XS abort: unhandled exception
# Break: C: xsDebugger!
  #0: (host)
  #1: @anonymous-2193
  #2: (anonymous-649) /home/pi/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/node-red-mcu/setupwifi.js:47
  #3: (anonymous-812) /home/pi/Projects/moddable/examples/network/wifi/wificonnection/wificonnection.js:95
  #4: (host)
# Disconnected from "main"
```
Since it has an "undefined" at nodered.js at line 1247, I assume the #broker is undefined:

[image]

Which means (line 1243) it cannot find my config node, although it is available in my flow editor. 
This is how my simple flow looks like:

[image]

Got a good tip from @ralphwetzel, to look in the working directory.  This is the flows.json file I found there:
```
[
  {
    "id": "ccdecb8b3a9bc97b",
    "type": "tab",
    "label": "WT32-ETH01",
    "disabled": false,
    "info": "",
    "env": []
  },
  {
    "id": "45e2f062e7beecdd",
    "type": "inject",
    "z": "ccdecb8b3a9bc97b",
    "name": "Every 5 seconds",
    "props": [
      {
        "p": "payload"
      }
    ],
    "repeat": "5",
    "crontab": "",
    "once": false,
    "onceDelay": 0.1,
    "topic": "",
    "payload": "",
    "payloadType": "date",
    "x": 510,
    "y": 240,
    "wires": [
      [
        "b0ce969590fa53d8",
        "816c5104e0b07c7e"
      ]
    ]
  },
  {
    "id": "b0ce969590fa53d8",
    "type": "debug",
    "z": "ccdecb8b3a9bc97b",
    "name": "debug 23",
    "active": true,
    "tosidebar": true,
    "console": false,
    "tostatus": false,
    "complete": "false",
    "statusVal": "",
    "statusType": "auto",
    "x": 720,
    "y": 240,
    "wires": []
  },
  {
    "id": "816c5104e0b07c7e",
    "type": "mqtt out",
    "z": "ccdecb8b3a9bc97b",
    "name": "",
    "topic": "waterput_koer",
    "qos": "",
    "retain": "",
    "respTopic": "",
    "contentType": "",
    "userProps": "",
    "correl": "",
    "expiry": "",
    "broker": "ed237c6b.40d1b",
    "x": 740,
    "y": 320,
    "wires": []
  },
  {
    "id": "f34c827151a897e5",
    "type": "catch",
    "z": "ccdecb8b3a9bc97b",
    "name": "",
    "scope": null,
    "uncaught": false,
    "x": 520,
    "y": 440,
    "wires": [
      [
        "5961fb2400ca8868"
      ]
    ]
  },
  {
    "id": "5961fb2400ca8868",
    "type": "debug",
    "z": "ccdecb8b3a9bc97b",
    "name": "Errors WT32-ETH01 waterput koer",
    "active": true,
    "tosidebar": true,
    "console": false,
    "tostatus": false,
    "complete": "payload",
    "targetType": "msg",
    "statusVal": "",
    "statusType": "auto",
    "x": 780,
    "y": 440,
    "wires": []
  }
]
```
As you can see, ***the json of the config node itself (node id "ed237c6b.40d1b") is NOT concluded in this flows.json file***. 

I had created this flow from scratch, to be used with node-red-mcu.  Except from the config node which already existed from quite some time.  As you can see at the "." in the middle of the broker config node id, which was the old way of creating node id's in Node-RED.  

So I did an extra test: ***when I create a new config node via my wt32-eth01 flow, then the json of that new config node is correctly added to my flow.json file***.  And then I see the messages arriving smoothly in my MQTT broker!  

So it would be nice if you could have a look if you could have a look why existing config nodes are not exported to the flows.json file in the working directory.

BTW. There is one difference with a 'normal' MQTT-Out node: the status is ***not*** *"connected"*, as soon as it is connected to the mqtt broker.  Not sure if you can do anything about that...

Thanks !!!

Bart

ralphwetzel, 2023-03-16:
Hi @bartbutenaers 
How did you create this flows.json?
If by the plugin, then it's - most probably - a plugin issue ...

ralphwetzel, 2023-03-16:
> So I did an extra test: _**when I create a new config node via my wt32-eth01 flow, then the json of that new config node is correctly added to my flow.json file**_. And then I see the messages arriving smoothly in my MQTT broker!

That's the result I get here as well trying to simulate your issue. There needs to be something different in the config you have.

Could you please post the flow definition as NR exports it: Menu -> Export -> Current flow -> JSON -> Copy to Clipboard. This flow definition should hold the data for the config node...

phoddie, 2023-03-16:
Thanks, @ralphwetzel. It didn't understand why the broker node was missing from the exported flow.

> BTW. There is one difference with a 'normal' MQTT-Out node: the status is not "connected", as soon as it is connected to the mqtt broker. Not sure if you can do anything about that...

@bartbutenaers – I'd be happy to take a look at this. But, I'm not 100% sure I fllow and it is kind of a separate issue. The MQTT broker does try to propagate status information to connected MQTT-In and MQTT-Out nodes. But, it is very possible that behavior isn't a perfect match with Node-RED (or there is a bug). Would you mind opening a separate issue with additional detail so that I can understand the difference you are seeing?

closed 2023-03-16 by phoddie

bartbutenaers, 2023-03-16:
@ralphwetzel: you nailed it!  I can confirm that it now works, which is nice because now I can easily reuse my existing config nodes.  Very useful fix.  Thanks a lot!!

@phoddie: I created the issue in your repo, because I thought the flow was generated by your code...
Yes you are are right that it is a completely separate issue, so I will create one.

phoddie, 2023-03-16:
> I created the issue in your repo, because I thought the flow was generated by your code....

No problem at all. The source of the problem is not always obvious.

> ...it is a completely separate issue, so I will create one.

Thank you.