phoddie/node-red-mcu #104: MQTT-Out node does not show status "connected"

bartbutenaers, 2023-03-16:
Hi @phoddie,

When I add an MQTT-Out node in a standard Node-RED flow, the node will show status *"connected"* as soon as the node has managed to connect to the specified MQTT broker:

[image]

However when I deploy the same flow - using Ralph's sidebar - to my wt32-eth01 device, then ***NO STATUS*** is being displayed below the MQTT-Out node (although the mcu is sending messages to my MQTT broker at that moment):

[image]

It would be nice if this could be added, similar to like the following status is already being displayed inside my wt32-eth01 flow:

[image]

Although while writing this, I start doubting whether this needs to be fixed in this repo or in the sidebar repo from @ralphwetzel.    
I mean that I am not sure anymore now whether the status updates are the responsibility of Ralph's proxy perhaps?

Thanks!!
Bart

phoddie, 2023-03-16:
Hey, @bartbutenaers – that's clear now. Thank you.

I will check that the MCU runtime is sending the status messages (I see the code to do this behavior is present, but I'll step through it and check the output to be sure it is working as expected).

phoddie, 2023-03-16:
@bartbutenaers – I used a simple test flow with an MQTT-Out node with id `6a2d91009192a70e`. It appears to be sending status messages back through the usual channel. The plug-in can receive these and forward them to the Node-RED Editor.

[image]

Perhaps there's something unexpected about the messages?

phoddie, 2023-03-16:
@bartbutenaers – I don't believe you should need to do anything special to allow the status messages to propagate through. The message trace shown in the my image above should be exactly what is received by the MCU plugin's proxy. Let's see what @ralphwetzel uncovers. 

As an aside – it took me a while to understand the status mechanism in Node-RED. It is used both for debugging in the Node-RED Editor and for receiving signals by nodes. Like many things in Node-RED, they are obvious/intuitive to experienced users but really not to someone seeing them the first time. The messages are formatted more for the debugging purpose, so I imagine that was their initial use. Anyway... as I've understood it better, I've also gotten more consistent about implementing support for them in the MCU implementation of nodes. For example, the MCU sensor nodes shows Connected or Disconnected status for sensors based on whether the node was able to connect to the sensor hardware.

bartbutenaers, 2023-03-16:
@phoddie,
Don't want to go too much off-topic, because off course this is an issue and not a discussion about statusses in general ;-)

However after all the nodes I have developed, there are still things about statusses that I keep doubting about during developing:
+ If an error occurs in the node, do I show a red error status.  Of course you see visually immediately that an error occurs.  But suppose that node doesn't receive an input message anymore for a couple of weeks.  Do you remove the error status after a while?  Because after two weeks perhaps the problem is not relevant anymore, because the root cause meanwhile has been updated.
+ If your node is streaming live data, you can show a static status text "streaming".   But suppose there is a gap in the stream of multiple minutes.  Do you remove the "streaming" status again, because no data is arriving anymore.  Or do you keep the "streaming" status, because the stream is still open.
+ And so on...

So there is not really one single way to go.  And every node developer has its own way of working, which makes it sometimes a bit hard to quickly interpret a flow based on the statusses....

But I will stop now about it ;-)

phoddie, 2023-03-17:
> So there is not really one single way to go. And every node developer has its own way of working, which makes it sometimes a bit hard to quickly interpret a flow based on the statusses....

Understood. That's kind of reassuring. My preference is not to invent yet another style, which is why I'm trying to discern patterns in the existing implementations. 

> But I will stop now about it ;-)

Me too! (Seems like a good topic for a presentation or a discussion)

ralphwetzel, 2023-03-17:
> However when I deploy the same flow - using Ralph's sidebar - to my wt32-eth01 device, then _**NO STATUS**_ is being displayed below the MQTT-Out node (although the mcu is sending messages to my MQTT broker at that moment):
> 
> [image]

Hi @bartbutenaers 
Sorry - I can't confirm what you report.
Are you sure, the MCU was (logically) connected to the editor?
As you're using xsbug-log: You can confirm this if you see the debug messages coming in as `<info>` data in the console monitor.

bartbutenaers, 2023-03-19:
Morning @ralphwetzel,
Sorry for the delay, I have a severe lack of free time lately due to circumstances...

Well I have to admit that it is not easy to test this in "very" limited time, because I constantly get that other error *"address already in use :::5004"*.  So I will first need to try to digest that [other](https://github.com/Moddable-OpenSource/moddable/issues/1063) discussion this evening.   I should have some more time to digg into the technical details, because at the moment I am bit lost with this...

phoddie, 2023-03-28:
The root cause here appears to be that the `-x` option was temporarily disabled in `mcconfig`. That was resolved yesterday in Moddable-OpenSource/moddable#1072.

closed 2023-03-28 by phoddie