phoddie/node-red-mcu #115: Is there a better way to reboot the I2C device and wait for it to come back up?

NextWorldLab, 2023-04-30:
Hello

With Arduino, the following code waits for a restart, but how can this be achieved with a Node-RED MCU?

```
  /* Reset */
  write8(BNO055_SYS_TRIGGER_ADDR, 0x20);  //device reset
  delay(30);
  while (read8(BNO055_CHIP_ID_ADDR) != BNO055_ID) {
    delay(10);
  }
  delay(50);
```
Now I can't read with I2CIn and get an error.
If Timeout can be selected, it may become possible to tranrate.

[image]

Is there a better way?

Thank,

NextWorldLab, 2023-04-30:
Try it with Catch Node.
Thank,

closed 2023-04-30 by NextWorldLab

NextWorldLab, 2023-04-30:
Catch Node is no good either.

[image]

NextWorldLab, 2023-04-30:
Extend the delay time and execute for the time being. .

phoddie, 2023-04-30:
The Catch node is triggered if there is an I2C error. I've confirmed that works.

<details>
<summary>I2C In with Catch</summary>

```json
[
    {
        "id": "b0d0dbc2ece07c0e",
        "type": "tab",
        "label": "sensor example",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": true
        }
    },
    {
        "id": "234e93bd172e1ec6",
        "type": "inject",
        "z": "b0d0dbc2ece07c0e",
        "name": "",
        "props": [],
        "repeat": "",
        "crontab": "",
        "once": true,
        "onceDelay": 0.1,
        "topic": "",
        "_mcu": {
            "mcu": true
        },
        "x": 210,
        "y": 200,
        "wires": [
            [
                "5324c751ef9eb63b"
            ]
        ]
    },
    {
        "id": "5324c751ef9eb63b",
        "type": "mcu_i2c_in",
        "z": "b0d0dbc2ece07c0e",
        "name": "",
        "options": {
            "bus": "default",
            "address": "0x48"
        },
        "command": "0x12",
        "bytes": 1,
        "moddable_manifest": {
            "include": "$(NODEREDMCU)/nodes/mcu/i2c/manifest.json"
        },
        "_mcu": {
            "mcu": true
        },
        "x": 390,
        "y": 200,
        "wires": [
            [
                "66c97e28603bcfb4"
            ]
        ]
    },
    {
        "id": "66c97e28603bcfb4",
        "type": "debug",
        "z": "b0d0dbc2ece07c0e",
        "name": "debug",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": true
        },
        "x": 550,
        "y": 300,
        "wires": []
    },
    {
        "id": "70894837354cf742",
        "type": "catch",
        "z": "b0d0dbc2ece07c0e",
        "name": "i2c fail",
        "scope": [
            "5324c751ef9eb63b"
        ],
        "uncaught": false,
        "_mcu": {
            "mcu": true
        },
        "x": 210,
        "y": 300,
        "wires": [
            [
                "5324c751ef9eb63b",
                "66c97e28603bcfb4"
            ]
        ]
    }
]
```

</details>

But, if the read succeeds without an error but returns an invalid result, the Catch node cannot help.

In either case, you don't really need any Delay nodes. You can just route a message back to the I2C In node to retry. Because of the way Node-RED processes messages, it will never block other nodes. That would look like this:

[image]

A completely different solution:  you could implement an ECMA-419 Sensor driver for the BNO055 orientation sensor. It will allow you to write code exactly like your Arduino example. Then it can just work with the Sensor node.

NextWorldLab, 2023-04-30:
thank you
It may not work properly due to some kind of communication failure. This circuit does not have a pull-up resistor, so connect it and try again. (I checked with the information that Arduino works)

Thanks for suggesting his ECMA-419 sensor driver implementation for BNO055. I think the threshold is a little high for me.

phoddie, 2023-04-30:
> It may not work properly due to some kind of communication failure. This circuit does not have a pull-up resistor, so connect it and try again. (I checked with the information that Arduino works)

OK. Good luck. From my tests above, I'm reasonably sure that the Node-RED MCU runtime has the pieces you need to make this work. If I can help somehow, please let me know.

phoddie, 2023-04-30:
My colleague points out this note from the [Adafruit page](https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor/overview) about the BNO055.

> The BNO055 I2C implementation violates the I2C protocol in some circumstances. This causes it not to work well with certain chip families. It does not work well with Espressif ESP32, ESP32-S3, and NXP i.MX RT1011, and it does not work well with I2C multiplexers. Operation with SAMD51, RP2040, STM32F4, and nRF52840 is more reliable.

Maybe this is related to your communication failures? They note that RP2040 works. If you have a Raspberry Pi Pico available, you might try that.

NextWorldLab, 2023-05-01:
thank you for the advice.

There may be a problem with the part you pointed out.
I would like to check with RP2040.