phoddie/node-red-mcu #116: Cannot build when using NodeRED MCU Plugin(WIndows).

NextWorldLab, 2023-05-03:
I don't know if the ISSUE in the repository here is okay.

I can no longer build with the Node-RED MCU Plugin.
You can BUILD from the command line.
The error message is "### Error: '.\\nodes\mcu\i2c\manifest.json': manifest not found!"

[image]

```
3 May 18:38:04 - [info] Node-RED version: v3.0.2
3 May 18:38:04 - [info] Node.js  version: v18.13.0
3 May 18:38:04 - [info] Windows_NT 10.0.19045 x64 LE
3 May 18:38:05 - [info] Loading palette nodes
3 May 18:38:06 - [info] Node-RED MCU Edition Runtime Version: #ddba250
3 May 18:38:06 - [info] Node-RED MCU Edition Plugin  Version: v1.3.1
3 May 18:38:10 - [info] Moddable SDK Version: v3.9.0-0-ga3ea980 (x86)
3 May 18:38:10 - [info] *** node-red-mcu-plugin:
3 May 18:38:10 - [info] It looks as if a new platform option has been added.
3 May 18:38:10 - [info] Please raise an issue @ our GitHub repository, stating the following information:
3 May 18:38:10 - [info] > New platform: esp32/qtpys2
3 May 18:38:10 - [info] > New platform: esp32/s2mini
3 May 18:38:10 - [info] > New platform: pico/qt_trinkey
3 May 18:38:10 - [info] > New platform: pico/xiao_ili9341
3 May 18:38:12 - [info] Dashboard version 3.4.0 started at /ui
3 May 18:38:12 - [info] Settings file  : C:\Users\tauchi\.node-red\settings.js
3 May 18:38:12 - [info] Context store  : 'default' [module=memory]
3 May 18:38:12 - [info] User directory : \Users\tauchi\.node-red
3 May 18:38:12 - [warn] Projects disabled : editorTheme.projects.enabled=false
3 May 18:38:12 - [warn] Flows file name not set. Generating name using hostname.
3 May 18:38:12 - [info] Flows file     : \Users\tauchi\.node-red\flows_NotePC-SZ6.json
````

```JSON:flows.json
[
    {
        "id": "484f9b23d91d9343",
        "type": "tab",
        "label": "フロー 1",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": true
        }
    },
    {
        "id": "385d04315cfa1bad",
        "type": "inject",
        "z": "484f9b23d91d9343",
        "name": "",
        "props": [
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": true,
        "onceDelay": "0.1",
        "topic": "",
        "_mcu": {
            "mcu": true
        },
        "x": 170,
        "y": 140,
        "wires": [
            [
                "9254c86c1b5de259"
            ]
        ]
    },
    {
        "id": "9254c86c1b5de259",
        "type": "mcu_i2c_in",
        "z": "484f9b23d91d9343",
        "name": "",
        "options": {
            "bus": "default",
            "address": "0x28"
        },
        "command": "0",
        "bytes": 1,
        "moddable_manifest": {
            "include": "$(NODEREDMCU)/nodes/mcu/i2c/manifest.json"
        },
        "_mcu": {
            "mcu": true
        },
        "x": 330,
        "y": 140,
        "wires": [
            [
                "a043a135084839cd"
            ]
        ]
    },
    {
        "id": "a043a135084839cd",
        "type": "debug",
        "z": "484f9b23d91d9343",
        "name": "debug 4",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": true
        },
        "x": 500,
        "y": 140,
        "wires": []
    }
]

````

manifest.json automatically generated by plugin
```
{
  "build": {
    "MODULES": "c:\\pjt\\moddable\\modules",
    "MCUROOT": "C:\\Users\\tauchi\\.node-red\\node_modules\\@ralphwetzel\\node-red-mcu-plugin\\node-red-mcu",
    "REDNODES": "C:\\Users\\tauchi\\AppData\\Roaming\\npm\\node_modules\\node-red\\node_modules\\@node-red\\nodes"
  },
  "include": [
    "$(MCUROOT)/manifest_host.json",
    "$(MCUROOT)/./nodes/mcu/i2c/manifest.json"
  ],
  "modules": {
    "*": [
      "./main",
      {
        "source": "./flows",
        "transform": "nodered2mcu"
      }
    ],
    "~": []
  },
  "preload": [
    "flows"
  ],
  "config": {
    "noderedmcu": {
      "editor": true
    }
  }
}
```
Is it a problem of Plugin?

thanks,

ralphwetzel, 2023-05-03:
> Is it a problem of Plugin?

I don't think so:

> ```json
>         "moddable_manifest": {
>             "include": "$(NODEREDMCU)/nodes/mcu/i2c/manifest.json"
>         },
> ```

in combination with 

https://github.com/phoddie/node-red-mcu/blob/ddba2504b4907a93814d41f6b86103593af1359a/manifest_runtime.json#L2-L4

creates `.//nodes/mcu/i2c/manifest.json`

... which is the erroneous path definition you can spot in your screencopy of the debug panel: It begins with `.//nodes/...`

[image]

phoddie, 2023-05-03:
@NextWorldLab – thank for the report and apologies for the trouble. And thank you @ralphwetzel for the quick attention.

It looks like the problem may be in how mcconfig is processing the relative definitions in the BUILD section of the manifest. That's a new feature and it looks like I may have gotten the handling of slashes wrong.

I don't have a Windows machine available right now to test, but I tried a quick fix. Maybe you can try running it? If not, we can test on Windows in the next day too. Here's the change. In $MODDABLE/tools/mcmanifest.js change the implementation of the parseBuild function to this:

<details>
<summary>parseBuild</summary>

```js
	parseBuild(platform) {
		let properties = platform.build;
		if (properties) {
			for (let name in properties) {
				let value = properties[name];
				if (typeof value == "string") {
					const dotSlash = "." + this.slash;
					value = this.resolveVariable(value);
					if (value.startsWith(dotSlash)) {
						const path = this.resolveDirectoryPath(dotSlash);
						if (path) {
							if (dotSlash == value)
								value = path;
							else
								value = path + value.slice(1);
						}
					}
					this.environment[name] = value;
				}
				else
					this.environment[name] = value;
			}
		}
	}
```

</details>

Then you'll need to rebuild tools:

```text
cd %MODDABLE%\build\makefiles\win
build
```

Then you can try rebuilding with the MCU plugin.

NextWorldLab, 2023-05-04:
@phoddie san, @ralphwetzel  san
Thank you for your response.

It worked fine by modifying parseBuild.

Thanks,

closed 2023-05-04 by NextWorldLab

phoddie, 2023-05-04:
Thank you for the quick confirmation! We will get that fix into a release as soon as we can.