phoddie/node-red-mcu #117: Dashboard manifest.json not found

NextWorldLab, 2023-05-06:
Hello
It may be the same as the other day, but I can't find the dashboard manifest.json. File exists.

Windows (Same error on Ubuntu (WSL environment).
[image]

```
6 May 12:21:04 - [info] Node-RED version: v3.0.2
6 May 12:21:04 - [info] Node.js  version: v18.13.0
6 May 12:21:04 - [info] Windows_NT 10.0.19045 x64 LE
6 May 12:21:06 - [info] Loading palette nodes
6 May 12:21:07 - [info] Node-RED MCU Edition Runtime Version: #93de634
6 May 12:21:07 - [info] Node-RED MCU Edition Plugin  Version: v1.3.1
6 May 12:21:12 - [info] Moddable SDK Version: v3.9.1-0-g70faab7 (x86)
6 May 12:21:12 - [info] *** node-red-mcu-plugin:
6 May 12:21:12 - [info] It looks as if a new platform option has been added.
6 May 12:21:12 - [info] Please raise an issue @ our GitHub repository, stating the following information:
6 May 12:21:12 - [info] > New platform: esp32/qtpys2
6 May 12:21:12 - [info] > New platform: esp32/s2mini
6 May 12:21:12 - [info] > New platform: pico/qt_trinkey
6 May 12:21:12 - [info] > New platform: pico/xiao_ili9341
6 May 12:21:18 - [info] Dashboard version 3.4.0 started at /ui
6 May 12:21:18 - [info] Settings file  : C:\Users\tauchi\.node-red\settings.js
6 May 12:21:18 - [info] Context store  : 'default' [module=memory]
6 May 12:21:18 - [info] User directory : \Users\tauchi\.node-red
6 May 12:21:18 - [warn] Projects disabled : editorTheme.projects.enabled=false
6 May 12:21:18 - [warn] Flows file name not set. Generating name using hostname.
6 May 12:21:18 - [info] Flows file     : \Users\tauchi\.node-red\flows_NotePC-SZ6.json
```

flows.json
```
[
    {
        "id": "2790209210860b70",
        "type": "tab",
        "label": "フロー 1",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": true
        }
    },
    {
        "id": "645031eb42957dd9",
        "type": "inject",
        "z": "2790209210860b70",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "_mcu": {
            "mcu": true
        },
        "x": 220,
        "y": 180,
        "wires": [
            [
                "6771b44add5492ea"
            ]
        ]
    },
    {
        "id": "6771b44add5492ea",
        "type": "ui_text",
        "z": "2790209210860b70",
        "group": "05a278c5cb8a11b0",
        "order": 0,
        "width": 0,
        "height": 0,
        "name": "",
        "label": "text",
        "format": "{{msg.payload}}",
        "layout": "row-spread",
        "className": "",
        "_mcu": {
            "mcu": true
        },
        "x": 410,
        "y": 180,
        "wires": []
    },
    {
        "id": "05a278c5cb8a11b0",
        "type": "ui_group",
        "name": "tb",
        "tab": "3db2d4310e9d80cc",
        "order": 1,
        "disp": false,
        "width": "6",
        "collapse": false
    },
    {
        "id": "3db2d4310e9d80cc",
        "type": "ui_tab",
        "name": "home",
        "icon": "dashboard",
        "disabled": false,
        "hidden": false
    }
]
```

manifest.json automatically generated by Node-RED MCU Plugin
```
{
  "build": {
    "MODULES": "c:\\pjt\\moddable\\modules",
    "MCUROOT": "C:\\Users\\tauchi\\.node-red\\node_modules\\@ralphwetzel\\node-red-mcu-plugin\\node-red-mcu",
    "REDNODES": "C:\\Users\\tauchi\\AppData\\Roaming\\npm\\node_modules\\node-red\\node_modules\\@node-red\\nodes"
  },
  "include": [
    "$(MCUROOT)/manifest_host.json",
    "$(MCUROOT)/./nodes/dashboard/manifest.json"
  ],
  "modules": {
    "*": [
      "./main",
      {
        "source": "./flows",
        "transform": "nodered2mcu"
      }
    ],
    "~": []
  },
  "preload": [
    "flows"
  ],
  "config": {
    "noderedmcu": {
      "editor": true
    }
  }
}
```

thanks ,

phoddie, 2023-05-06:
This is caused by a bug in mcconfig that is triggered in the plug-in build environment, which builds outside the Node-RED MCU repository. That's perfectly reasonable, but was overlooked. The fix is in `mcmanifest.js`. Change this...

```js
					else if (nodeTypes[node.type])
						this.parseManifest(source, {include: nodeTypes[node.type], directory: this.resolveVariable("$(NODEREDMCU)")});
```

To this:

```js
					else if (nodeTypes[node.type]) {
						const save = this.currentDirectory;
						this.currentDirectory = this.resolveVariable("$(NODEREDMCU)");
						this.parseManifest(source, {include: nodeTypes[node.type], directory: this.currentDirectory});
						this.currentDirectory = save;
					}
```

...and then rebuild Moddable SDK tools. 

There is a workaround, which may be a good idea for the near term. The workaround  is to change $NODEREDMCU/node_types.json to use absolute paths. For example, replace this

```
	"ui_button": "./nodes/dashboard/manifest.json",
```

...with this...

```
	"ui_button": "$(NODEREDMCU)/nodes/dashboard/manifest.json",
```
...for all the relative paths.

I successfully tested that with the MCU plug-in. The advantage is that users only need to update their Node-RED MCU repository, not the Moddable SDK.

@ralphwetzel – your input here would be helpful before deciding whether to deploy the workaround. Thank you!

ralphwetzel, 2023-05-06:
[This code fragment](https://github.com/Moddable-OpenSource/moddable/blob/70faab72f45651ebc841f3e1072c0446b3db39a2/tools/mcmanifest.js#L1865-L1870) 

```
	flows.forEach((node, i) => {
		if (node.moddable_manifest)
			this.parseManifest(source, {...node.moddable_manifest, directory: this.currentDirectory});
		else if (nodeTypes[node.type])
			this.parseManifest(source, {include: nodeTypes[node.type], directory: this.resolveVariable("$(NODEREDMCU)")});
	});
```

looks like you're always _magically_ including the manifests for nodes listed in `node_types.json` - which I think is a quite clever change.
Consequentially - if you confirm - the plugin doesn't need to include the (path to the) manifests for those nodes anymore in the generated `manifest.json`.
I could remove the few lines that handle this topic & release a new version - if this helps to solve the issue easier.

NextWorldLab, 2023-05-07:
@phoddie san, @ralphwetzel san

Thank you for your response.

Thank you for telling me how to fix it.
For the time being, I modified mcmanifest.js and it worked correctly.

Thanks,

phoddie, 2023-05-07:
@ralphwetzel – thanks for taking a look. You are correct -- I've been working towards having the build include the required node manifests, to only build the nodes used in given flow. I eventually realized that the `node_types.json` database that Patrick added for you for the MCU plug-in could be used to resolve the built-in nodes.

I'll commit an update today to `node_types.json` to use absolute paths temporarily to work around the mcconfig bug. In my testing, that works the MCU plug-in. If you can confirm that and then update the plug-in, that would be great.

ralphwetzel, 2023-05-07:
@phoddie I've updated the plugin. It will no longer include the manifests for nodes in `node_types.json`. Thus it's save to change this file now...

phoddie, 2023-05-07:
Thanks, @ralphwetzel. I've committed the change to `node_types.json`.

I also committed a change that splits the MQTT nodes out of the core into a module. They rely on the magic include mechanism to build.

NextWorldLab, 2023-05-09:
@phoddie san, @ralphwetzel san

I've verified that it works fine with a github clone.

thank you.

closed 2023-05-09 by NextWorldLab