phoddie/node-red-mcu #118: MQTT does not work

NextWorldLab, 2023-05-12:
hello

MQTT does not work on Windows.(Haven't tried other platforms.)

Error Message(xsbug)
```
C:\Users\tauchi\.node-red\node_modules\@ralphwetzel\node-red-mcu-plugin\node-red-mcu\nodes\network\mqtt\mqttnodes.js (282) # Break: MQTTOutNode.prototype.onStart: get configFlowID: undefined variable!
```

Node-RED
```
Welcome to Node-RED
===================

12 May 13:19:42 - [info] Node-RED version: v3.0.2
12 May 13:19:42 - [info] Node.js  version: v18.13.0
12 May 13:19:42 - [info] Windows_NT 10.0.22621 x64 LE
12 May 13:19:43 - [info] Loading palette nodes
12 May 13:19:43 - [info] Node-RED MCU Edition Runtime Version: #40f5369
12 May 13:19:43 - [info] Node-RED MCU Edition Plugin  Version: v1.3.2
12 May 13:19:44 - [info] Moddable SDK Version: v3.9.1-0-g70faab7 (x86)
12 May 13:19:44 - [info] *** node-red-mcu-plugin:
12 May 13:19:44 - [info] It looks as if a new platform option has been added.
12 May 13:19:44 - [info] Please raise an issue @ our GitHub repository, stating the following information:
12 May 13:19:44 - [info] > New platform: esp32/qtpys2
12 May 13:19:44 - [info] > New platform: esp32/s2mini
12 May 13:19:44 - [info] > New platform: pico/qt_trinkey
12 May 13:19:44 - [info] > New platform: pico/xiao_ili9341
12 May 13:19:44 - [info] Dashboard version 3.3.1 started at /ui
12 May 13:19:44 - [info] Settings file  : C:\Users\tauchi\.node-red\settings.js
12 May 13:19:44 - [info] Context store  : 'default' [module=memory]
12 May 13:19:44 - [info] User directory : \Users\tauchi\.node-red
12 May 13:19:44 - [warn] Projects disabled : editorTheme.projects.enabled=false
12 May 13:19:44 - [warn] Flows file name not set. Generating name using hostname.
12 May 13:19:44 - [info] Flows file     : \Users\tauchi\.node-red\flows_t-frontier.json
12 May 13:19:44 - [info] Server now running at http://127.0.0.1:1880/
12 May 13:19:44 - [warn]
```
Flows.json
```
[
    {
        "id": "daed785743b85aaf",
        "type": "tab",
        "label": "フロー 2",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": true
        }
    },
    {
        "id": "6869687472d7e171",
        "type": "inject",
        "z": "daed785743b85aaf",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "_mcu": {
            "mcu": true
        },
        "x": 320,
        "y": 140,
        "wires": [
            [
                "7165b148a79f9105"
            ]
        ]
    },
    {
        "id": "7165b148a79f9105",
        "type": "mqtt out",
        "z": "daed785743b85aaf",
        "name": "",
        "topic": "MCU/Topic1",
        "qos": "",
        "retain": "",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "5bc5facb368b5484",
        "_mcu": {
            "mcu": true
        },
        "x": 650,
        "y": 220,
        "wires": []
    },
    {
        "id": "5bc5facb368b5484",
        "type": "mqtt-broker",
        "name": "",
        "broker": "mqtt.beebotte.com",
        "port": "1883",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": "",
        "_mcu": {
            "mcu": false
        }
    }
]
```

thanks,

phoddie, 2023-05-12:
Thanks for the report. This is a result of moving the MQTT nodes from the Node-RED MCU core to a separate model. It looks like an import was missing. I've committed a fix. Please let me know if that works for you now.

NextWorldLab, 2023-05-12:
@phoddie  san

Thank you for your quick response.
It worked fine.

thank you.

closed 2023-05-12 by NextWorldLab