phoddie/node-red-mcu #130: Link error with esp32/esp32c6_cdc

NW-Lab, 2025-04-19:
hello

I can run Moddable Examples/hellowworld with esp32/esp32c6_cdc without any issues, but I get errors on the MCU.

environment
Windows10
Node.js  version: v20.11.1
Node-RED MCU Edition Runtime Version: #f17e3ab
Moddable SDK Version: v5.6.0-0-g2945c5d (x86)
ESP-IDF  v5.4.1 (Sorry if this is the cause of the problem.)

flows.json
```
[
    {
        "id": "8419c3b9beebe939",
        "type": "tab",
        "label": "フロー 8",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": true
        }
    },
    {
        "id": "b6bae28be30cb6fc",
        "type": "inject",
        "z": "8419c3b9beebe939",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "5",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "_mcu": {
            "mcu": true
        },
        "x": 170,
        "y": 100,
        "wires": [
            [
                "af97bf40396e6221"
            ]
        ]
    },
    {
        "id": "af97bf40396e6221",
        "type": "debug",
        "z": "8419c3b9beebe939",
        "name": "debug 4",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": true
        },
        "x": 400,
        "y": 100,
        "wires": []
    }
]
```

Part of the error
```
FAILED: xs_esp32.elf
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\Espressif\tools\riscv32-esp-elf\esp-14.2.0_20241119\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C6=0 -Wl,--Map=C:/pjt/moddable/build/tmp/esp32/esp32c6_cdc/debug/node-red-mcu/xsProj-esp32c6/build/xs_esp32.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32c6.peripherals.ld -T esp32c6.rom.ld -T esp32c6.rom.api.ld -T esp32c6.rom.rvfp.ld -T esp32c6.rom.wdt.ld -T esp32c6.rom.systimer.ld -T esp32c6.rom.version.ld -T esp32c6.rom.phy.ld -T esp32c6.rom.coexist.ld -T esp32c6.rom.net80211.ld -T esp32c6.rom.pp.ld -T esp32c6.rom.newlib.ld -T esp32c6.rom.newlib-normal.ld -T esp32c6.rom.heap.ld -T memory.ld -T sections.ld @CMakeFiles\xs_esp32.elf.rsp -o xs_esp32.elf && cd ."
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: xs_esp32c6.a(digitalbank.c.o): in function `digitalISR':
c:\pjt\moddable\modules\io\digital\esp32/digitalbank.c:353:(.iram1.0+0x44): undefined reference to `__atomic_fetch_add_1'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: xs_esp32c6.a(digitalbank.c.o): in function `digitalDeliver':
c:\pjt\moddable\modules\io\digital\esp32/digitalbank.c:369:(.text.digitalDeliver+0x16): undefined reference to `__atomic_fetch_sub_1'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: xs_esp32c6.a(digitalbank.c.o): in function `xs_digitalbank_destructor':
c:\pjt\moddable\modules\io\digital\esp32/digitalbank.c:286:(.text.xs_digitalbank_destructor+0x5a): undefined reference to `__atomic_fetch_sub_1'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: xs_esp32c6.a(serial.c.o): in function `serialDeliver':
c:\pjt\moddable\modules\io\serial\esp32/serial.c:409:(.text.serialDeliver+0x1a): undefined reference to `__atomic_fetch_sub_1'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: xs_esp32c6.a(serial.c.o): in function `_uart_disable_intr_mask':
c:\pjt\moddable\modules\io\serial\esp32/serial.c:468:(.iram1.mod+0x84): undefined reference to `__atomic_fetch_add_1'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: xs_esp32c6.a(serial.c.o): in function `xs_serial_destructor':
c:\pjt\moddable\modules\io\serial\esp32/serial.c:247:(.text.xs_serial_destructor+0x76): undefined reference to `__atomic_fetch_sub_1'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

thank you.

phoddie, 2025-04-19:
GCC atomics aren't available on ESP32 family RISC-V MCUs. But, we have code in place that **should** account for that. I'll take a look.

> ESP-IDF v5.4.1 (Sorry if this is the cause of the problem.)

Thanks for noting this. I wouldn't expect it to be the source of the problem, but good to know.

NW-Lab, 2025-04-19:
thank you for your reply.

It's a problem with the RISC-V MPU. I understand.
I will try it with another MPU.

Thank you.

mkellner, 2025-04-21:
Please try this patch:

```
diff --git a/tools/mcconfig/nmake.esp32.mk b/tools/mcconfig/nmake.esp32.mk
index 0d1133f36..bbf7c91b4 100644
--- a/tools/mcconfig/nmake.esp32.mk
+++ b/tools/mcconfig/nmake.esp32.mk
@@ -437,7 +437,8 @@ C_COMMON_FLAGS = -c -Os -g \

 !IF "$(ESP_ARCH)"=="riscv"
 C_COMMON_FLAGS = $(C_COMMON_FLAGS) \
-       -march=rv32imc
+       -march=rv32imc \
+       -D_NO_ATOMICS
 !ELSE
 C_COMMON_FLAGS = $(C_COMMON_FLAGS) \
        -mlongcalls \

```

NW-Lab, 2025-04-22:
Thanks for the patch. This fixed the problem.

It seems that the ESP32C6 doesn't work on Mac or Linux either. I think you may have the same problem.

Thank  you.