phoddie/node-red-mcu #131: Serial2xsbug Connection Error

tesanders, 2025-05-07:
I've been able to compile, build, flash, and run the "MODDABLE/examples/helloworld" app on a "xiao_esp32s3_sense" module.  When I attempt to run a simple MCU Edition flow (inject>debug) on the same module, the app is successfully built and flashed onto the module, but the following error message is generated and the app never starts:
    >~/Projects/moddable/tools/serial2xsbug/serial2xsbug_lin.c:116  Connection refused

Environment Info:
    OS:  Ubuntu 24.04LTS
    Node-RED:  v4.0.5
    NodeJS:      v22.11.0
    Node-RED MCU Edition Runtime:  version: #f17e3ab
    Node-RED MCU Edition Plug-in:    version: v1.5.3
    MODDABLE SDK Version:  v5.6.0-0-g2945c5d
    ESP-IDF:   esp-idf-v5.4

phoddie, 2025-05-08:
Welcome, @tesanders.

There were some issues connecting to ESP32 devices with a built-in USB port, like the ESP32-S3, on macOS and Linux. Working with @mshioji, we were able to find a solution. The fixes will be included in Moddable SDK 5.7.0 which should be released in the coming days.

tesanders, 2025-05-08:
Thank you for the follow-up!  I'm looking forward to the SDK 5.7.0 release